gatecat/nextpnr-xilinx #4: SIGBUS when running netpnr-xilinx

fallen, 2019-12-24:
I don't know if the project is mature enough for someone to report issues yet but here is an issue I've just had:
```

(gdb) run
Starting program: /home/fallen/dev/nextpnr-xilinx/nextpnr-xilinx --chipdb ../../xc7a35t.bin --xdc arty.xdc --json top.json --write top_routed.json --fasm top.fasm
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Info: Importing module top
Info: Rule checker, verifying imported design
Info: Checksum: 0x63e7a2ba

Info: Packing constants..
Info: Inserting IO buffers..

Program received signal SIGBUS, Bus error.
0x00005555555be6e4 in nextpnr_xilinx::Arch::locInfo<nextpnr_xilinx::BelId> (this=0x55555b1c8b30, id=<synthetic pointer>...)
    at /home/fallen/dev/nextpnr-xilinx/xilinx/arch.h:1504
1504	        return chip_info->tile_types[chip_info->tile_insts[id.tile].type];
(gdb) p chip_info
$1 = (const nextpnr_xilinx::ChipInfoPOD *) 0x7ffff57d2d54
(gdb) bt
#0  0x00005555555be6e4 in nextpnr_xilinx::Arch::locInfo<nextpnr_xilinx::BelId> (this=0x55555b1c8b30, id=<synthetic pointer>...)
    at /home/fallen/dev/nextpnr-xilinx/xilinx/arch.h:1504
#1  nextpnr_xilinx::Arch::getBelName (this=this@entry=0x55555b1c8b30, bel=...) at /home/fallen/dev/nextpnr-xilinx/xilinx/arch.h:668
#2  0x00005555555b8dcd in nextpnr_xilinx::BaseCtx::nameOfBel (this=0x55555b1c8b30, bel=...) at /home/fallen/dev/nextpnr-xilinx/common/nextpnr.cc:215
#3  0x00005555556923f7 in nextpnr_xilinx::XC7Packer::pack_io (this=this@entry=0x7fffffffd2c0) at /home/fallen/dev/nextpnr-xilinx/xilinx/pack_io_xc7.cc:259
#4  0x0000555555667f62 in nextpnr_xilinx::Arch::pack (this=0x55555b1c8b30) at /home/fallen/dev/nextpnr-xilinx/xilinx/pack.cc:845
#5  0x00005555555a4ddd in nextpnr_xilinx::CommandHandler::executeMain (this=this@entry=0x7fffffffdac0, ctx=std::unique_ptr<nextpnr_xilinx::Context> = {...})
    at /home/fallen/dev/nextpnr-xilinx/common/command.cc:309
#6  0x00005555555a6f44 in nextpnr_xilinx::CommandHandler::exec (this=0x7fffffffdac0) at /home/fallen/dev/nextpnr-xilinx/common/command.cc:384
#7  0x000055555559ac64 in main (argc=<optimized out>, argv=<optimized out>) at /home/fallen/dev/nextpnr-xilinx/xilinx/main.cc:119
(gdb) p chip_info->tile_types
$2 = {offset = -21018125}

```

I can provide the top.json: [top.json.zip](https://github.com/daveshah1/nextpnr-xilinx/files/3998136/top.json.zip)

top.json was generated with yosys like that:
```
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_cpu.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_adder.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_logic_op.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_dcache.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_addsub.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog top.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_dtlb.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_decoder.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_itlb.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_shifter.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_mc_arithmetic.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_instruction_unit.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_interrupt.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_multiplier.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_load_store_unit.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_icache.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_ram.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_dp_ram.v
read_verilog -I/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog /home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_debug.v
attrmap -tocase keep -imap keep="true" keep=1 -imap keep="false" keep=0 -remove keep=0
synth_xilinx -flatten -nobram -arch xcup -top top
write_json top.json
```

daveshah1, 2019-12-27:
Thanks for reporting, I can reproduce with the attached netlist and will investigate.

daveshah1, 2019-12-27:
Should be fixed now, the problem was related to unconstrained IO placement for xc7

closed 2019-12-27 by daveshah1

daveshah1, 2019-12-27:
Incidentally, the attached design is a good test for `router2` (feel free to give it a test), it reduces the total runtime from 18 minutes to 5 minutes.

fallen, 2020-02-07:
I've run the tool on this design and it took 48 min FYI.

Here is the complete log:
```

Info: Importing module top
Info: Rule checker, verifying imported design
Info: Checksum: 0x913daeb7

Info: Packing constants..
Info: Inserting IO buffers..
Info:     Created 7 IOB33_INBUF_EN cells from:
Info:              7x IBUF
Info:     Created 2 IOB33_OUTBUF cells from:
Info:              2x OBUF
Info:     Created 9 PAD cells from:
Info:              9x PAD
Info: Preparing clocking...
Info: Packing constants..
Info: Packing PLLs...
Info: Packing global buffers...
Info: Packing MUX[789]s..
Info: Packing carries..
Info:    Grouped 491 MUXCYs and 427 XORCYs into 26 chains.
Info:    Blasted 0 non-chain MUXCYs and 0 non-chain XORCYs to soft logic
Info: Packing LUTs..
Info:     Created 6312 SLICE_LUTX cells from:
Info:            635x LUT1
Info:           1831x LUT2
Info:            610x LUT3
Info:            385x LUT4
Info:            386x LUT5
Info:           2465x LUT6
Info: Packing DRAM..
Info:    Transformed 16 tied-low DRAM address inputs to be tied-high
Info: Packing BRAM..
Info: Packing flipflops..
Info:     Created 1801 SLICE_FFX cells from:
Info:           1801x FDRE
Info:     Created 2568 SELMUX2_1 cells from:
Info:           2413x MUXF7
Info:            155x MUXF8
Info: Constrained 1427 LUTFF pairs.

Info: Annotating ports with timing budgets for target frequency 12.00 MHz
Info: Checksum: 0x32fa4e2d

Info: Device utilisation:
Info: 	          SLICE_LUTX: 10340/65200    15%
Info: 	           SLICE_FFX:  1801/65200     2%
Info: 	              CARRY4:   139/ 8150     1%
Info: 	          PSEUDO_GND:     1/18055     0%
Info: 	          PSEUDO_VCC:     1/18055     0%
Info: 	               HARD0:     0/ 1980     0%
Info: 	   RAMB18E1_RAMB18E1:     0/  150     0%
Info: 	   FIFO18E1_FIFO18E1:     0/   75     0%
Info: 	RAMBFIFO36E1_RAMBFIFO36E1:     0/   75     0%
Info: 	   RAMB36E1_RAMB36E1:     0/   75     0%
Info: 	     DSP48E1_DSP48E1:     0/  120     0%
Info: 	                 PAD:     9/  730     1%
Info: 	       IOB33M_OUTBUF:     0/  120     0%
Info: 	       IOB33S_OUTBUF:     0/  120     0%
Info: 	        IOB33_OUTBUF:     2/  250     0%
Info: 	     IOB33M_INBUF_EN:     0/  120     0%
Info: 	     IOB33S_INBUF_EN:     0/  120     0%
Info: 	      IOB33_INBUF_EN:     7/  250     2%
Info: 	IOB33M_TERM_OVERRIDE:     0/  120     0%
Info: 	IOB33S_TERM_OVERRIDE:     0/  120     0%
Info: 	 IOB33_TERM_OVERRIDE:     0/  250     0%
Info: 	       PULL_OR_KEEP1:     0/  490     0%
[... 330 lines cut ...]
Info:  legend: * represents 6 endpoint(s)
Info:          + represents [1,6) endpoint(s)
Info: [ 71775,  72312) |+
Info: [ 72312,  72849) |*****+
Info: [ 72849,  73386) |**+
Info: [ 73386,  73923) |+
Info: [ 73923,  74460) |********+
Info: [ 74460,  74997) |**************+
Info: [ 74997,  75534) |***************************+
Info: [ 75534,  76071) |******************+
Info: [ 76071,  76608) |**********************+
Info: [ 76608,  77145) |************************************************************ 
Info: [ 77145,  77682) |********************************************+
Info: [ 77682,  78219) |***************************************************+
Info: [ 78219,  78756) |**********************************+
Info: [ 78756,  79293) |***********+
Info: [ 79293,  79830) |***********************+
Info: [ 79830,  80367) |**********************************************+
Info: [ 80367,  80904) |*******************************+
Info: [ 80904,  81441) |****************************************+
Info: [ 81441,  81978) |*************************************************+
Info: [ 81978,  82515) |*************+
Info: Running post-routing legalisation...

```

commit 5bf1a175 (2019-12-27): xc7: Fix placement of unconstrained IO

Fixes #4