gatecat/nextpnr-xilinx #11: Exception when --chipdb argument is not supplied

Xiretza, 2020-02-04:
Minor detail, but `nextpnr-xilinx` crashes with an exception when the `--chipdb` command-line argument is not supplied. On 95d7a1d:

```
$ gdb ./nextpnr-xilinx
GNU gdb (GDB) 8.3.1
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./nextpnr-xilinx...
(gdb) r
Starting program: /home/xiretza/dev/nextpnr-xilinx/build/nextpnr-xilinx 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
/usr/lib/../share/gcc-9.2.0/python/libstdcxx/v6/xmethods.py:731: SyntaxWarning: list indices must be integers or slices, not str; perhaps you missed a comma?
  refcounts = ['_M_refcount']['_M_pi']
nextpnr-xilinx -- Next Generation Place and Route (Version 95d7a1dc)

General options:
  -h [ --help ]             show help
  -v [ --verbose ]          verbose output
  -q [ --quiet ]            quiet mode, only errors and warnings displayed
  -l [ --log ] arg          log file, all log messages are written to this file
                            regardless of -q
  --debug                   debug output
  -f [ --force ]            keep running after errors
  --gui                     start gui
  --gui-no-aa               disable anti aliasing (use together with --gui 
                            option)
  --run arg                 python file to execute instead of default flow
  --pre-pack arg            python file to run before packing
  --pre-place arg           python file to run before placement
  --pre-route arg           python file to run before routing
  --post-route arg          python file to run after routing
  --json arg                JSON design file to ingest
  --write arg               JSON design file to write
  --seed arg                seed value for random number generator
  -r [ --randomize-seed ]   randomize seed value for random number generator
  --placer arg              placer algorithm to use; available: sa, heap; 
                            default: heap
  --slack_redist_iter arg   number of iterations between slack redistribution
  --cstrweight arg          placer weighting for relative constraint 
                            satisfaction
  --starttemp arg           placer SA start temperature
  --placer-budgets          use budget rather than criticality in placer timing
                            weights
  --pack-only               pack design only without placement or routing
  --no-route                process design without routing
  --no-place                process design without placement
  --no-pack                 process design without packing
  --ignore-loops            ignore combinational loops in timing analysis
  -V [ --version ]          show version
  --test                    check architecture database integrity
  --freq arg                set target frequency for design in MHz
  --timing-allow-fail       allow timing to fail in design
  --no-tmdriv               disable timing-driven placement
  --sdf arg                 SDF delay back-annotation file to write
  --sdf-cvc                 enable tweaks for SDF file compatibility with the 
                            CVC simulator

Architecture specific options:
  --chipdb arg              name of chip database binary
  --xdc arg                 XDC-style constraints file
  --fasm arg                fasm bitstream file to write

terminate called after throwing an instance of 'boost::wrapexcept<boost::bad_any_cast>'
  what():  boost::bad_any_cast: failed conversion using boost::any_cast

Program received signal SIGABRT, Aborted.
0x00007ffff6037f25 in raise () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff6037f25 in raise () from /usr/lib/libc.so.6
#1  0x00007ffff6021897 in abort () from /usr/lib/libc.so.6
#2  0x00007ffff63f281d in __gnu_cxx::__verbose_terminate_handler () at /build/gcc/src/gcc/libstdc++-v3/libsupc++/vterminate.cc:95
#3  0x00007ffff63ff4da in __cxxabiv1::__terminate (handler=<optimized out>) at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:47
#4  0x00007ffff63ff537 in std::terminate () at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:57
#5  0x00007ffff63ff78e in __cxxabiv1::__cxa_throw (obj=obj@entry=0x555555c20930, tinfo=tinfo@entry=0x555555be0770 <typeinfo for boost::wrapexcept<boost::bad_any_cast>>, 
    dest=dest@entry=0x5555557e8900 <boost::wrapexcept<boost::bad_any_cast>::~wrapexcept()>) at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:95
#6  0x00005555557bc410 in boost::throw_exception<boost::bad_any_cast> (e=...) at /usr/include/boost/exception/exception.hpp:217
#7  0x00005555557ef222 in boost::any_cast<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&> (operand=...) at /usr/include/c++/9.2.0/bits/exception.h:63
#8  0x000055555592642b in boost::any_cast<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&> (operand=...) at /usr/include/boost/any.hpp:295
#9  boost::program_options::variable_value::as<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (this=<optimized out>)
    at /usr/include/boost/program_options/variables_map.hpp:69
#10 UspCommandHandler::createContext (this=<optimized out>, values=...) at /home/xiretza/dev/nextpnr-xilinx/xilinx/main.cc:68
#11 0x00005555557e5dd7 in nextpnr_xilinx::CommandHandler::exec (this=this@entry=0x7fffffffdf60) at /home/xiretza/dev/nextpnr-xilinx/common/command.cc:385
#12 0x00005555557d8e75 in main (argc=<optimized out>, argv=<optimized out>) at /home/xiretza/dev/nextpnr-xilinx/xilinx/main.cc:88
```

daveshah1, 2020-02-04:
Coincidentally I just fixed this in the new `xilinx-upstream` branch (which is my WIP getting nextpnr-xilinx upstream-ready). I'll backport the patch.

Xiretza, 2020-02-04:
Hah, that's great! I was actually testing from there, seems it just hasn't been pushed yet. I'm always on the fence about opening issues vs. asking on IRC vs. just waiting a few days while development is this fast-paced, what's your preference there?

closed 2020-02-18 by Xiretza