gatecat/nextpnr-xilinx #17: X-Ray Database incompatible to nextpnr-xilinx FASM

Haini, 2020-05-14:
Updating the project x-ray database to commit [405c3bc720466a9963f1c302de5f0c6b4ee18073](https://github.com/SymbiFlow/prjxray-db) yields the following error when we want to encode a bitstream wiht nextpnr-xilinx generated FASM.  
 
```
Traceback (most recent call last):
  File "/home/prjxray/utils/fasm2frames.py", line 306, in <module>
    main()
  File "/home/prjxray/utils/fasm2frames.py", line 302, in main
    emit_pudc_b_pullup=args.emit_pudc_b_pullup)
  File "/home/prjxray/utils/fasm2frames.py", line 179, in run
    assembler.parse_fasm_filename(filename_in, extra_features=extra_features)
  File "/home/prjxray/prjxray/fasm_assembler.py", line 180, in parse_fasm_filename
    raise FasmLookupError('\n'.join(missing_features))
prjxray.fasm_assembler.FasmLookupError: Segment DB RIOB33, key RIOB33.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.IN_ONLY not found from line 'RIOB33_X43Y75.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.IN_ONLY'
Segment DB RIOB33, key RIOB33.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW not found from line 'RIOB33_X43Y61.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW'
Segment DB RIOB33, key RIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW not found from line 'RIOB33_X43Y59.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW'
Segment DB RIOB33, key RIOB33.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW not found from line 'RIOB33_X43Y55.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW'
Segment DB RIOB33, key RIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW not found from line 'RIOB33_X43Y53.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW'
Segment DB RIOB33, key RIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW not found from line 'RIOB33_X43Y61.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW'
Segment DB RIOB33, key RIOB33.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW not found from line 'RIOB33_X43Y57.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW'
Segment DB RIOB33, key RIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW not found from line 'RIOB33_X43Y55.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW'
Segment DB RIOB33, key RIOB33.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW not found from line 'RIOB33_X43Y51.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW'
Segment DB RIOB33, key RIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW not found from line 'RIOB33_X43Y63.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW'
Segment DB RIOB33, key RIOB33.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW not found from line 'RIOB33_X43Y59.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW'
Segment DB RIOB33, key RIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW not found from line 'RIOB33_X43Y57.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW'
Segment DB RIOB33, key RIOB33.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW not found from line 'RIOB33_X43Y53.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW'
Segment DB LIOB33, key LIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.IN_ONLY not found from line 'LIOB33_X0Y125.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.IN_ONLY'
Segment DB LIOB33, key LIOB33.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.IN_ONLY not found from line 'LIOB33_X0Y123.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.IN_ONLY'
Segment DB LIOB33, key LIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.IN_ONLY not found from line 'LIOB33_X0Y123.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.IN_ONLY'
Segment DB LIOB33, key LIOB33.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.IN_ONLY not found from line 'LIOB33_X0Y121.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.IN_ONLY'
```

Turns out some elements were renamed:

```
diff --git a/artix7/segbits_riob33.db b/artix7/segbits_riob33.db
index 88ceb7d..b07295c 100644
--- a/artix7/segbits_riob33.db
+++ b/artix7/segbits_riob33.db
 RIOB33.IOB_Y0.LVCMOS15.DRIVE.I12 38_64 38_112 !38_118 !38_126 39_65 39_113 !39_117 !39_119 !39_125 39_127
-RIOB33.IOB_Y0.LVCMOS15.DRIVE.I16 38_64 38_112 38_118 38_126 39_65 39_113 39_117 !39_119 !39_125 39_127
 RIOB33.IOB_Y0.LVCMOS15_LVCMOS18_LVCMOS25.DRIVE.I4 38_64 38_112 38_118 !38_126 39_65 39_113 !39_117 !39_119 39_125 39_127
+RIOB33.IOB_Y0.LVCMOS15_SSTL15.DRIVE.I16_I_FIXED 38_64 38_112 38_118 38_126 39_65 39_113 39_117 !39_119 !39_125 39_127
 RIOB33.IOB_Y0.LVCMOS18.DRIVE.I12_I8 38_64 38_112 !38_118 38_126 39_65 39_113 !39_117 39_119 39_125 !39_127
 RIOB33.IOB_Y0.LVCMOS18.DRIVE.I16 38_64 38_112 !38_118 38_126 39_65 39_113 !39_117 !39_119 39_125 39_127
 RIOB33.IOB_Y0.LVCMOS18.DRIVE.I24 38_64 38_112 !38_118 38_126 39_65 39_113 39_117 39_119 !39_125 !39_127
@@ -33,9 +33,9 @@ RIOB33.IOB_Y0.LVCMOS33_LVTTL.DRIVE.I4 38_64 !38_112 38_118 !38_126 39_65 !39_117
 RIOB33.IOB_Y0.LVCMOS33_LVTTL.DRIVE.I12_I8 38_64 !38_112 !38_118 38_126 39_65 !39_117 39_119 39_125 !39_127
 RIOB33.IOB_Y0.LVCMOS33_LVTTL.DRIVE.I12_I16 38_64 !38_112 !38_118 38_126 39_65 !39_117 !39_119 39_125 39_127
 RIOB33.IOB_Y0.SSTL135.DRIVE.I_FIXED 38_64 38_112 38_118 38_126 39_65 39_113 39_117 39_119 !39_125 39_127
-RIOB33.IOB_Y0.SSTL135.IN !38_86 !39_85 39_87
-RIOB33.IOB_Y0.SSTL135.IN_DIFF 38_86 !39_85 39_87
-RIOB33.IOB_Y0.SSTL135.SLEW.FAST !38_106 38_110 39_105 39_107 39_109 39_111
+RIOB33.IOB_Y0.SSTL135_SSTL15.IN !38_86 !39_85 39_87
+RIOB33.IOB_Y0.SSTL135_SSTL15.IN_DIFF 38_86 !39_85 39_87
+RIOB33.IOB_Y0.SSTL135_SSTL15.SLEW.FAST !38_106 38_110 39_105 39_107 39_109 39_111
 RIOB33.IOB_Y1.IBUFDISABLE.I 39_45
 RIOB33.IOB_Y1.IN_TERM.NONE !38_04 !38_06 !39_05 !39_07
 RIOB33.IOB_Y1.IN_TERM.UNTUNED_SPLIT_40 38_04 38_06 39_05 39_07
@@ -52,14 +52,14 @@ RIOB33.IOB_Y1.LVCMOS12.DRIVE.I4 !38_00 38_02 !38_08 !38_10 38_14 38_62 39_01 39_
 RIOB33.IOB_Y1.LVCMOS12.DRIVE.I12 38_00 !38_02 !38_08 !38_10 38_14 38_62 39_01 !39_09 39_15 39_63
 RIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18.IN !38_40 38_42 39_41
 RIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL.SLEW.FAST !38_16 !38_18 !38_20 !38_22 !39_17 !39_21
-RIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.IN_ONLY !38_00 38_02 38_08 !38_10 38_14 !38_62 !39_01 39_09 !39_15 !39_63
-RIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135.SLEW.SLOW !38_16 38_18 !38_20 38_22 39_17 39_21
-RIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_SSTL135.STEPDOWN 38_32
+RIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135_SSTL15.IN_ONLY !38_00 38_02 38_08 !38_10 38_14 !38_62 !39_01 39_09 !39_15 !39_63
+RIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135_SSTL15.SLEW.SLOW !38_16 38_18 !38_20 38_22 39_17 39_21
+RIOB33.IOB_Y1.LVCMOS12_LVCMOS15_LVCMOS18_SSTL135_SSTL15.STEPDOWN 38_32
 RIOB33.IOB_Y1.LVCMOS12_LVCMOS25.DRIVE.I8 !38_00 !38_02 38_08 !38_10 38_14 38_62 !39_01 !39_09 39_15 39_63
 RIOB33.IOB_Y1.LVCMOS15.DRIVE.I8 38_00 38_02 38_08 !38_10 38_14 38_62 39_01 !39_09 39_15 39_63
[... 90 lines cut ...]
--- 
I guess that `RIOB33.IOB_Y1.LVCMOS15_SSTL15.DRIVE.I16_I_FIXED` needs an additional entry in the following code section

```fasm.cc:672
        if (is_output) {
            if (iostandard == "LVCMOS33" || iostandard == "LVTTL")
                write_bit("LVCMOS33_LVTTL.DRIVE.I12_I16");
            if (iostandard == "SSTL135")
                write_bit("SSTL135.DRIVE.I_FIXED");
+           if (iostandard == "SSTL15")
+                write_bit("LVCMOS15_SSTL15.DRIVE.I16_I_FIXED");
            if (slew == "SLOW")
                ...
```

---
`RIOB33.IOB_Y0.SSTL135_SSTL15.IN_DIFF` probably needs to be added around here (I didn't find any usage of `SSTL135.IN_DIFF` or `SSTL135.IN_DIFF`, so I hardcoded it): 

```fasm.cc:696
        if (is_input && diff) {
-            write_bit(iostandard + ".IN_DIFF");
+           write_bit("SSTL135_SSTL15.IN_DIFF")
```
---
Also: What implications does this change in naming have: Does it just mean that the same bit is responsible for some more configurations? I know that SSTL is Stub-Series Terminated Logic, so it would make sense that the same bit has influence on different voltages.

daveshah1, 2020-05-14:
Thanks for catching this, I'll have a more detailed look soon. This seems to be because prjxray has gained SSTL15 support, and indeed it merges the names of shared config bits. I think all thse changes are fine but I'll need to test them first.

Haini, 2020-05-14:
Hi and thanks for the fast explanation!

I think you will need to pay attention >when< a SSTL15 Bit should be inserted as I am not very experienced with the code base / architecture yet. The proposed changes are just best guesses.

Haini, 2020-05-15:
Nice, closing this issue then.

closed 2020-05-15 by Haini