gatecat/nextpnr-xilinx #21: Incorrect FASM lines related to SERDES

acomodi, 2020-07-31:
Due to a recent update in the prjxray-db, the ISERDES fasm lines have changed. Currently nextpnr seems to produce incorrect fasm lines when SERDESes are introduced in a design:

Error example:
```
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y39.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y39.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_X43Y45.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y45.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y45.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_X43Y33.ILOGIC_Y0.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y33.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y33.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_X43Y29.ILOGIC_Y0.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y29.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y29.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_X43Y35.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y35.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y35.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_X43Y27.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y27.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y27.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_X43Y33.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y33.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y33.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_X43Y27.ILOGIC_Y0.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y27.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y27.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_X43Y45.ILOGIC_Y0.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y45.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y45.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3_TBYTETERM, key RIOI3_TBYTETERM.ILOGIC_Y0.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_TBYTETERM_X43Y37.ILOGIC_Y0.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3_TBYTETERM, key RIOI3_TBYTETERM.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_TBYTETERM_X43Y37.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3_TBYTETERM, key RIOI3_TBYTETERM.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_TBYTETERM_X43Y37.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_X43Y41.ILOGIC_Y0.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y41.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y41.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_X43Y47.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y47.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y47.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_X43Y39.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y39.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y39.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_X43Y41.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y41.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y41.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_X43Y29.ILOGIC_Y1.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y29.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y29.ILOGIC_Y1.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.DATA_WIDTH.W8 not found from line 'RIOI3_X43Y25.ILOGIC_Y0.ISERDES.DATA_WIDTH.W8'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY not found from line 'RIOI3_X43Y25.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.Z_MEMORY'
Segment DB RIOI3, key RIOI3.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY not found from line 'RIOI3_X43Y25.ILOGIC_Y0.ISERDES.INTERFACE_TYPE.NOT_MEMORY'
```

The error is produced during the bitstream generation step.

This prjxray PR changed the ISERDES fasm lines: https://github.com/SymbiFlow/prjxray/pull/1406

daveshah1, 2020-08-01:
@acomodi  As far as I can tell prjxray-db is in the intermediate step of the broken features-starting-with-numbers still and hasn't actually been updated with your last PR yet?

acomodi, 2020-08-03:
I think prjxray-db has been actually updated, this is the [last merge-commit](https://github.com/SymbiFlow/prjxray-db/commit/002032fa463ab11026afde28c7a91e8a2156bb10) containing the changes to ISERDES fasm features.

daveshah1, 2020-08-03:
No, it isn't. That is the broken dot-number style. The latest prjxray moves to a dot-W-number format. (also note that merge-commit significantly predates the most recent X-ray PR.)

acomodi, 2020-08-03:
Right, my mistake, I was referring to this PR in prjxray, but pasted the wrong one: https://github.com/SymbiFlow/prjxray/pull/978. This one changes the ISERDES features

from:
```
ISERDES.DATA_WIDTH.<data_width>
ISERDES.INTERFACE_TYPE.<interface_type>
```

to:
```
ISERDES.<interface_type>.<data_width>
```

daveshah1, 2020-08-03:
But I tried that and it seems to just result in an error in the FASM tooling?

daveshah1, 2020-08-05:
New database now pushed so I've fixed the nextpnr side. Thanks for reporting!

commit 2425dde6 (2020-08-05): xilinx: Update ISERDES FASM generation

Fixes #21
closed 2020-08-05 by daveshah1