gatecat/nextpnr-xilinx #22: ERROR: IDELAYE2 'IDELAYE2' has disconnected IDATAIN input

fallen, 2020-09-19:
I am struggling to understand what's happening here:

```
$ gdb --args $NEXTPNR_DIR/nextpnr-xilinx -v --chipdb $NEXTPNR_DIR/xilinx/xc7a35t.bin --xdc arty.xdc --json arty.json --write top_routed.json --fasm arty.fasm --debug
(gdb) b log_error
Breakpoint 1 at 0x798c0: file /home/fallen/dev/nextpnr-xilinx/common/log.cc, line 164.
(gdb) run
Starting program: /home/fallen/dev/nextpnr-xilinx/nextpnr-xilinx -v --chipdb /home/fallen/dev/nextpnr-xilinx//xilinx/xc7a35t.bin --xdc arty.xdc --json arty.json --write top_routed.json --fasm arty.fasm --debug
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Info: Importing module arty
Info: processing output port serial_tx
Info: processing input port serial_rx
Info: processing input port cpu_reset
Info: processing input port clk100
Info: processing output port eth_ref_clk
Info: processing output port ddram_a[0]
Info: processing output port ddram_a[1]
Info: processing output port ddram_a[2]
Info: processing output port ddram_a[3]
Info: processing output port ddram_a[4]
Info: processing output port ddram_a[5]
Info: processing output port ddram_a[6]
Info: processing output port ddram_a[7]
Info: processing output port ddram_a[8]
Info: processing output port ddram_a[9]
Info: processing output port ddram_a[10]
Info: processing output port ddram_a[11]
Info: processing output port ddram_a[12]
Info: processing output port ddram_a[13]
Info: processing output port ddram_ba[0]
Info: processing output port ddram_ba[1]
Info: processing output port ddram_ba[2]
Info: processing output port ddram_ras_n
Info: processing output port ddram_cas_n
Info: processing output port ddram_we_n
Info: processing output port ddram_cs_n
Info: processing output port ddram_dm[0]
Info: processing output port ddram_dm[1]
Info: processing output port ddram_clk_p
Info: processing output port ddram_clk_n
Info: processing output port ddram_cke
Info: processing output port ddram_odt
Info: processing output port ddram_reset_n
Info: processing output port user_led
Info: processing inout port ddram_dq[0]
Info: processing inout port ddram_dq[1]
Info: processing inout port ddram_dq[2]
Info: processing inout port ddram_dq[3]
Info: processing inout port ddram_dq[4]
Info: processing inout port ddram_dq[5]
Info: processing inout port ddram_dq[6]
Info: processing inout port ddram_dq[7]
Info: processing inout port ddram_dq[8]
Info: processing inout port ddram_dq[9]
Info: processing inout port ddram_dq[10]
Info: processing inout port ddram_dq[11]
Info: processing inout port ddram_dq[12]
Info: processing inout port ddram_dq[13]
Info: processing inout port ddram_dq[14]
Info: processing inout port ddram_dq[15]
Info: processing inout port ddram_dqs_p[0]
Info: processing inout port ddram_dqs_p[1]
Info: processing inout port ddram_dqs_n[0]
Info: processing inout port ddram_dqs_n[1]
Info: Rule checker, verifying imported design
Info: Checksum: 0x812f73fc

Info: ignoring unsupported XDC command 'set_false_path' (on line 306)
Info: ignoring unsupported XDC command 'set_false_path' (on line 308)
[... 155 lines cut ...]
            "REFCLK_FREQUENCY": "200.000000",
            "SIGNAL_PATTERN": "DATA"
          },
          "attributes": {
            "module_not_derived": 1,
            "src": "arty.v:11051"
          },
          "port_directions": {
            "DATAOUT": "output",
            "IDATAIN": "input",
            "LD": "input"
          },
          "connections": {
            "DATAOUT": [ 13042 ],
            "IDATAIN": [ 14522 ],
            "LD": [ 3145 ]
          }
        },

```

I'm attaching the files: [issue.zip](https://github.com/daveshah1/nextpnr-xilinx/files/5250439/issue.zip)

Any idea?
Thanks!

fallen, 2020-09-21:
It seems this port gets disconnected by this piece of code:

```
(gdb) bt
#0  0x00005555555cafad in nextpnr_xilinx::disconnect_port (ctx=<optimized out>, 
    cell=0x55556000b430, port_name=...)
    at /home/fallen/dev/nextpnr-xilinx/common/design_utils.cc:118
#1  0x000055555568456a in nextpnr_xilinx::XilinxPacker::flush_cells (
    this=this@entry=0x7fffffffd130) at /usr/include/c++/9/bits/unique_ptr.h:360
#2  0x00005555556c68aa in nextpnr_xilinx::XC7Packer::pack_io (
    this=this@entry=0x7fffffffd130)
    at /home/fallen/dev/nextpnr-xilinx/xilinx/pack_io_xc7.cc:277
#3  0x000055555569237a in nextpnr_xilinx::Arch::pack (this=0x55555cab5c90)
    at /home/fallen/dev/nextpnr-xilinx/xilinx/pack.cc:910
#4  0x00005555555bb774 in nextpnr_xilinx::CommandHandler::executeMain (
    this=0x7fffffffd9d0, ctx=std::unique_ptr<nextpnr_xilinx::Context> = {...})
    at /usr/include/c++/9/bits/unique_ptr.h:360
#5  0x00005555555bd91f in nextpnr_xilinx::CommandHandler::exec (this=0x7fffffffd9d0)
    at /usr/include/c++/9/bits/move.h:74
#6  0x00005555555b1fb9 in main (argc=<optimized out>, argv=<optimized out>)
    at /home/fallen/dev/nextpnr-xilinx/xilinx/main.cc:88
```

fallen, 2020-10-07:
Don't hesitate to ask me to add any more detail on the issue.
I can also give a try to some patch! :+1:

daveshah1, 2020-10-10:
Sorry, this got forgotten about. Its a familiar error, I will look into the detail tomorrow.

fallen, 2020-10-10:
Thanks a lot! :)

daveshah1, 2020-10-11:
Trying the json provided, I get
```
ERROR: Found an out-of-range integer parameter in the JSON file.
Please regenerate the input file with an up-to-date version of yosys.
```
are you sure that your yosys and nextpnr-xilinx are up to date?

fallen, 2020-10-11:
I'm on 
* Yosys release 0.9 (sha-1 1979e0b1f2482dbf0562f5116ab444280a377773)

But I think I was on an older nextpnr-xiling sha-1 .... I've just rebased on `origin/xilinx-upstream`
let me retry this.

daveshah1, 2020-10-11:
You will need to use Yosys from git master, not the last release.

fallen, 2020-10-11:
I can confirm that with up to date nextpnr-xilinx I get the same error as you.
So I need to use master branch of Yosys?

fallen, 2020-10-11:
Just saw your answer, thanks! Will try that.

fallen, 2020-10-11:
With up to date Yosys master branch build, I now get this error:
```
Info:     Created 3 DSP48E1_DSP48E1 cells from:
Info:              3x DSP48E1
ERROR: Clocked DSP48E1s are currently unsupported (while processing cell $flatten\lm32_cpu.\multiplier.$mul$/home/fallen/.local/lib/python3.6/site-packages/misoc-0.6.dev0-py3.6.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_multiplier.v:115$4898.sliceA.last.sliceB[0].mul, clocked by blinkysoc_clkout_buf0).
0 warnings, 1 error
```

the json file: [arty.json.zip](https://github.com/daveshah1/nextpnr-xilinx/files/5361215/arty.json.zip)

The json is generated via: 
```
attrmap -tocase keep -imap keep="true" keep=1 -imap keep="false" keep=0 -remove keep=0
synth_xilinx -flatten -arch xc7 -top arty
write_json arty.json
```

daveshah1, 2020-10-11:
Try adding `scratchpad -set xilinx_dsp.multonly 1` to the yosys script before `synth_xilinx`

fallen, 2020-10-11:
It worked!
Thanks!
What is it doing?

daveshah1, 2020-10-11:
That disables use of the DSP registers and other fancy features by Yosys - at the time I did the DSP support in nextpnr-xilinx prjxray only had limited DSP support, although things may have improved since then.

fallen, 2020-10-11:
Thanks a lot!
Closing this ticket :)

closed 2020-10-11 by fallen