gatecat/nextpnr-xilinx #24: Instantiation of Xilinx PS hard IP

ilesser, 2020-10-10:
### Description
I am working  in PR #20 to add an example for the Ultra96 board.

The blinky example requires instantiating the PS (processing system) hard IP because the clock is obtained from there.

I created a blackbox module for the PS name `zynq_ultra_ps_e_0` in [here](https://github.com/ilesser/nextpnr-xilinx/blob/ultra96-example/xilinx/examples/ultra96/zynq_ultra_ps_e_0.v).

I am able to synthesize the design but when I call `nextpnr-xilinx` I get this error:

```
ERROR: Unable to place cell 'u_zynq_ultra_ps_e_0', no Bels remaining of type 'zynq_ultra_ps_e_0'
0 warnings, 1 error
```

### Question
Is there a way to tell the tool where to get the information for the IP?

I can save the IP information as a TCL file, how could I read in that information?

Does the tool support mapping to specific hard IPs?

If it doesn't how could I contribute to it?
I have plenty of experience with TCL and Python but little with C++.

daveshah1, 2020-10-10:
Hi, I will look at your other PR shortly. The way to use the PS is via a PS8 primitive- what you are attempting to use is a Xilinx IP core that wraps that. The IP core should be unencrypted Verilog so passing it to Yosys should work, you'll need to find all the files until you get down to a PS8 primitive.

ilesser, 2020-10-10:
I'll google how to do that. Thanks for the directions!

daveshah1, 2020-10-10:
If you have a simple use case (e.g. just wanting a single clock), then it might be easier to use the PS8 directly rather than try and make the Xilinx verilog work:

```verilog
    wire [3:0] plclk;
    PS8 ps_i(
        .PLCLK(plclk)
    );

    BUFG_PS bfg_i(
        .I(plclk[0]),
        .O(clk)
    );
```
is some code I have used on the zcu104 before.

ilesser, 2020-10-10:
So far the information I can find on the zynq_ultra_ps_e is all encripted I don't think we could read in the verilog.

I did found some .xci and .xml files with the ip information. But I don't think it would be possible to read them in.

I will give a try at your example.

By the way what is the PS8? Is it a PLL? I couldn't find it in neither the  UltraScale Architecture Libraries Guide [UG974](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2014_1/ug974-vivado-ultrascale-libraries.pdf) nor in the UltraScale Architecture Clocking Resources [UG572](https://www.xilinx.com/support/documentation/user_guides/ug572-ultrascale-clocking.pdf)

Thanks!

daveshah1, 2020-10-10:
Sadly, its not documented by Xilinx but PS8 is the low level interface to the Zynq block (processing system 8). It is what the Xilinx IP cores wrap.