gatecat/nextpnr-xilinx #27: ERROR: Unable to place cell: no Bels remaining of type 'MMCME2_ADV_MMCME2_ADV'

MJoergen, 2021-01-02:
I run the command
```
/home/mike/git/daveshah1/nextpnr-xilinx/nextpnr-xilinx --chipdb /home/mike/git/daveshah1/nextpnr-xilinx/xilinx/xc7a100t.bin --xdc src/queens_top.xdc --json queens.json --write queens_routed.json --fasm queens.fasm
```
but the command fails quickly with the error:
```
ERROR: Unable to place cell 'i_clk_wiz_0.u0.mmcm_adv_inst', no Bels remaining of type 'MMCME2_ADV_MMCME2_ADV'
```

I've used the Clocking Wizard in Vivado to generate a VHDL file that instantiates the MMCME2_ADV.

In the `json` file I see the instantiation as:
```
        "i_clk_wiz_0.u0.mmcm_adv_inst": {
          "hide_name": 0,
          "type": "MMCME2_ADV",
          "parameters": {
```
and the instance itself is declared as:
```
    "MMCME2_ADV": {
      "attributes": {
        "blackbox": "00000000000000000000000000000001",
        "src": "/usr/local/bin/../share/yosys/xilinx/cells_xtra.v:7933.1-8027.10"
      },
```

I've attached the `json` file.

[queens.json.gz](https://github.com/daveshah1/nextpnr-xilinx/files/5760469/queens.json.gz)

daveshah1, 2021-01-02:
Back when I was working on nextpnr-xilinx, prjxray only supported the PLL and not the MMCM. I don't know whether that is still true but it's how things are currently.

MJoergen, 2021-01-02:
Ok, thanks for the reply.