gatecat/nextpnr-xilinx #28: ERROR: Unable to place cell ..., no Bels remaining of type 'LDCE'

ildus, 2021-01-03:
`yosys` creates a file with LDCE primitives, apparently `nextpnr-xilinx` doesn't support them.
The command:

```
nextpnr-xilinx --chipdb /home/ildus/dev/nextpnr-xilinx/xilinx/xc7a35t.bin --xdc arty.xdc --json midi_ctrl.json --write midi_ctrl_routed.json --fasm midi_ctrl.fasm
```
Output:
```
ERROR: Unable to place cell '$auto$simplemap.cc:581:simplemap_dlatch$2082', no Bels remaining of type 'LDCE'
```

BrunoLevy, 2021-08-31:
In my own codebase, I noticed that LDCE cells are generated when there is a combinatorial block with a case() statement and a missing default value (then it generates a flipflop mapped to a LDCE). In my case it was due to errors in my own code.

racerxdl, 2022-02-14:
Me too, in my case it was a ALU that I had commented out the default case

```verilog
always @(*)
begin
  case (operation)
    ADD:                        result = X +  Y;
    SUB:                        result = X -  Y;
    OR:                         result = X |  Y;
    XOR:                        result = X ^  Y;
    AND:                        result = X &  Y;
    LesserThanUnsigned:         result = X <  Y;
    LesserThanSigned:           result = $signed(X) < $signed(Y);
    ShiftRightUnsigned:         result = X >> (Y[4:0]);
    ShiftRightSigned:           result = $signed(X) >>> (Y[4:0]);
    ShiftLeftUnsigned:          result = X << (Y[4:0]);
    ShiftLeftSigned:            result = $signed(X) <<< (Y[4:0]);
    GreaterThanOrEqualUnsigned: result = X >= Y;
    GreaterThanOrEqualSigned:   result = $signed(X) >= $signed(Y);
    Equal:                      result = X == Y;
    NotEqual:                   result = X != Y;
    default:                    result = 0; // This line commented out.
  endcase
end
```