gatecat/nextpnr-xilinx #32: segmentation fault

KiranKanchi, 2021-06-06:
Hello,

When i try to build the attosoc.sh, I get the following break:

4. Executing JSON backend.

End of script. Logfile hash: 080161d859, CPU: user 4.32s system 0.07s, MEM: 323.22 MB peak
Yosys 0.9+4081 (git sha1 2d10caab, clang 11.0.0-2 -fPIC -Os)
Time spent: 21% 27x read_verilog (1 sec), 13% 12x techmap (0 sec), ...
+ ../../../nextpnr-xilinx --chipdb ../../xc7a35t.bin --xdc arty.xdc --json attosoc.json --write attosoc_routed.json --fasm attosoc.fasm
./attosoc.sh: line 5: 25556 Segmentation fault      (core dumped) ../../../nextpnr-xilinx --chipdb ../../xc7a35t.bin --xdc arty.xdc --json attosoc.json --write attosoc_routed.json --fasm attosoc.fasm

Any pointers on what could be the problem ?

Note: ./bbasm xilinx/xc7a35t.bba xilinx/xc7a35t.bin --> i used bigendian as parameter here
I dont understand : you may also need to patch out the Vivado check for utils/environment.sh in Xray

Kind regards
KiranKanchi

gatecat, 2021-06-06:
> Note: ./bbasm xilinx/xc7a35t.bba xilinx/xc7a35t.bin --> i used bigendian as parameter here

are you really on big endian machine? `--l` is probably more appropriate...

KiranKanchi, 2021-06-06:
Let me restart with little endian. Thank you very much for kind support!

KiranKanchi, 2021-06-06:
you are right, little endian fixed the problem. 
Could you please elaborate on "you may also need to patch out the Vivado check for utils/environment.sh in Xray"

Thank you very much !
kind regards,
Kiran Kanchi

gatecat, 2021-06-06:
> Could you please elaborate on "you may also need to patch out the Vivado check for utils/environment.sh in Xray"

Remove everything from this line and beyond from `environment.sh` in prjxray: https://github.com/SymbiFlow/prjxray/blob/master/utils/environment.sh#L52

KiranKanchi, 2021-06-06:
worked! thank you so much!
Have a great day/evening!

Kind regards,
Kiran Kanchi

closed 2021-06-06 by KiranKanchi

gatecat, 2021-06-06:
No problem, I've just fixed some of the README issues so it's a bit clearer for anyone in the future.