gatecat/nextpnr-xilinx #35: `bbaexport.py` fails for `xc7a35t`

ajelinski, 2022-03-09:
The problem was introduced with the https://github.com/gatecat/nextpnr-xilinx/commit/565588a69ea95a52f7c7592f4ed81d9bef6cfb60 commit.
Basically, such a directory doesn't exist: `external/prjxray-db/artix7/xc7a35t/tilegrid.json`

See https://github.com/hdl/conda-eda/issues/191 for more details.

tmichalak, 2022-03-09:
It also seems that the [prjxray-db](https://github.com/gatecat/nextpnr-xilinx/tree/xilinx-upstream/xilinx/external) is heavily outdated.

jakubhladik1, 2022-11-06:
The bbaexport.py also fails for xc7z020clg400-1:

`python3 xilinx/python/bbaexport.py --device xc7z020clg400-1 --bba xilinx/xc7z020.bba
Traceback (most recent call last):
  File "/Users/jhladik/Documents/Development/HDL/zybo-playground/tools/nextpnr-xilinx/xilinx/python/bbaexport.py", line 350, in <module>
    main()
  File "/Users/jhladik/Documents/Development/HDL/zybo-playground/tools/nextpnr-xilinx/xilinx/python/bbaexport.py", line 31, in main
    d = import_device(args.device, xraydb_root, metadata_root)
  File "/Users/jhladik/Documents/Development/HDL/zybo-playground/tools/nextpnr-xilinx/xilinx/python/xilinx_device.py", line 462, in import_device
    with open(prjxray_root + "/" + fabricname + "/tilegrid.json") as gf:
FileNotFoundError: [Errno 2] No such file or directory: '/Users/jhladik/Documents/Development/HDL/zybo-playground/tools/nextpnr-xilinx/xilinx/python/../external/prjxray-db/zynq7/xc7z020clg400-1t/tilegrid.json'
`

There is no "t" in the FPGA part number so this splitting used by `bbaexport.py` won't work for Zynq-7000 family FPGAs/SoCs.
 
I had to change the `fabricname = name.split('t')[0] + "t"` to `fabricname = name`  to make it work.