gatecat/nextpnr-xilinx #57: Big incompatibility of the code structure with the actual NextPNR

zipotron, 2022-12-29:
Hello. I am trying to port the https://github.com/gatecat/nextpnr-xilinx code into NextPNR mainstream. Why? Well, NextPNR had some time ago a big code/arch refactor, unfortunately NextPNR-Xilinx project is not a fork of NextPNR, you just copy the code and start to develop on it, and NextPNR-Xilinx actually is based on the NextPNR version before of the big refactor. This means that all the Xilinx code developed for NextPNR is completely incompatible with the actual NextPNR. Is that huge the architecture differences that I am already working on this task (that I thought was easy) for two days and I am completely overload.
 I am working in this repo:
https://github.com/zipotron/nextpnr/tree/xilinx
I already did many changes, but I reach a point, that I need a opinion/tips/help from people familiar with the code, for example, my last commit, is the adaptation to the Arch Class to be derivative of BaseArch, and have some "function pointers" that I am not sure I did connect with the right functions.
Please, i need supervision. Any one can help me?
In my opinion is a shame let that project become obsolete, I am in touch with the owner, waiting for response too.
Thanks

gatecat, 2022-12-29:
Ultimately nextpnr-xilinx was an experiment and if I did it again there are definitely many things I would change, deduplicating the database for starters.

@mmicko is also working on this in #56 - and I have some outline for how I'd approach a new xilinx arch in https://github.com/YosysHQ/nextpnr/pull/816 but I've had other priorities recently

zipotron, 2022-12-30:
I came a couple of weeks late! Actually @mmicko have the work really advanced (mine is not compiling yet... I will keep following his work and help him if I can. Thanks!

closed 2022-12-30 by zipotron