gatecat/nextpnr-xilinx #65: FDRE_1 can't be written out to fasm file

jrsa, 2023-03-13:
`terminate called after throwing an instance of 'nextpnr_xilinx::assertion_failure'
  what():  Assertion failure: unsupported FF type (/home/jrsa/src/nextpnr-xilinx/xilinx/fasm.cc:536)`

The FF type in question turns out to be FDRE_1, after inspecting with gdb. It seems like the FDRE_1 is supported, from looking [https://github.com/gatecat/nextpnr-xilinx/blob/xilinx-upstream/xilinx/pack.cc#L230](elsewhere) in the code.

jrsa, 2023-03-19:
I thought for some reason that I could add `FDRE_1` [here](https://github.com/gatecat/nextpnr-xilinx/blob/634f1b5e86e233428c97f3153031415af4284c33/xilinx/fasm.cc#L519) to be handled the same way as `FDRE`. Looking [here](https://github.com/gatecat/nextpnr-xilinx/blob/xilinx-upstream/xilinx/pack.cc#L230) i get the impression that FDRE and FDRE_1 should be treated the same, except the _1 has the clock inverted. I did try `if (type == "FDRE" || type == "FDRE_1") {` but  I suspect that I'm still missing something necessary to make the inverted clock work.
As I understand it, this FF config will be used when you have a `always @(negedge clk)` which is unusual, in my case it used in the VexRiscV JTAG TAP.

jrsa, 2023-03-20:
I added some prints in `fasm.cc` and revealed that my inverted clock FF was placed in the same slice with other non-inverted ones:
```
Info: writing ffs in tile CLBLL_L_X2Y192 half 1
Info: IS_C_INVERTED 0
Info: IS_C_INVERTED 1
```

jschj, 2024-01-19:
Is this a bug? Did you solve it? I am having exactly the same problem.

jrsa, 2024-01-19:
@jschj I did not fix it, not enough knowledge about the internals of nextpnr to navigate it. I would be happy to work on it if I could be pointed in the right direction, perhaps by @gatecat. Are you trying to synthesize VexRiscV as well?

jschj, 2024-01-19:
Yes, also with the JTAG thing. I did some digging in the last hour and it seems like the problem starts in yosys because it places FlipFlops with normal and inverted clocks into the same slice which is I guess not allowed.

jschj, 2024-01-21:
@jrsa I think I found the problem and some dirty fix for now https://github.com/openXC7/nextpnr-xilinx/issues/21

jrsa, 2024-01-22:
@jschj out of curiosity what version of yosys are you using

jschj, 2024-01-22:
Yosys 0.37+15