gatecat/nextpnr-xilinx #77: Error when adding PLLE2_ADV in xc7a35

jtplaarj, 2023-11-14:
I have modified the blinky example for the arty board to add a PLLE2_ADV using the vivado clocking wizard.

The code looks like:

```verilog
module blinky (
    input  wire clk,
    output wire led
    );

    reg [24:0] r_count = 0;
    wire clk_out;

    always @(posedge(clk_out)) r_count <= r_count + 1;
    assign led = r_count[24];

  clk_wiz_0 clock_management_i
   (
    .clk_out1(clk_out),
    .clk_in1(clk)
);
endmodule
```
`clk_wiz_0` is obtained using the clocking wizard:

```verilog
`timescale 1ps/1ps

(* CORE_GENERATION_INFO = "clk_wiz_0,clk_wiz_v6_0_11_0_0,{component_name=clk_wiz_0,use_phase_alignment=true,use_min_o_jitter=false,use_max_i_jitter=false,use_dyn_phase_shift=false,use_inclk_switchover=false,use_dyn_reconfig=false,enable_axi=0,feedback_source=FDBK_AUTO,PRIMITIVE=PLL,num_out_clk=1,clkin1_period=10.000,clkin2_period=10.000,use_power_down=false,use_reset=false,use_locked=true,use_inclk_stopped=false,feedback_type=SINGLE,CLOCK_MGR_TYPE=NA,manual_override=false}" *)

module clk_wiz_0 
 (
  // Clock out ports
  output        clk_out1,
  // Status and control signals
  output        locked,
 // Clock in ports
  input         clk_in1
 );

  clk_wiz_0_clk_wiz inst
  (
  // Clock out ports  
  .clk_out1(clk_out1),
  // Status and control signals               
  .locked(locked),
 // Clock in ports
  .clk_in1(clk_in1)
  );

endmodule
```

The same for `clk_wiz_0_clk_wiz`:

```verilog
`timescale 1ps/1ps

module clk_wiz_0_clk_wiz 

 (// Clock in ports
  // Clock out ports
  output        clk_out1,
  // Status and control signals
  output        locked,
  input         clk_in1
 );
  // Input buffering
  //------------------------------------
wire clk_in1_clk_wiz_0;
wire clk_in2_clk_wiz_0;
  IBUF clkin1_ibufg
   (.O (clk_in1_clk_wiz_0),
    .I (clk_in1));

  // Clocking PRIMITIVE
  //------------------------------------

  // Instantiation of the MMCM PRIMITIVE
  //    * Unused inputs are tied off
  //    * Unused outputs are labeled unused

  wire        clk_out1_clk_wiz_0;
  wire        clk_out2_clk_wiz_0;
  wire        clk_out3_clk_wiz_0;
  wire        clk_out4_clk_wiz_0;
  wire        clk_out5_clk_wiz_0;
  wire        clk_out6_clk_wiz_0;
  wire        clk_out7_clk_wiz_0;

  wire [15:0] do_unused;
  wire        drdy_unused;
  wire        psdone_unused;
  wire        locked_int;
  wire        clkfbout_clk_wiz_0;
  wire        clkfbout_buf_clk_wiz_0;
  wire        clkfboutb_unused;
   wire clkout1_unused;
   wire clkout2_unused;
   wire clkout3_unused;
   wire clkout4_unused;
  wire        clkout5_unused;
  wire        clkout6_unused;
  wire        clkfbstopped_unused;
  wire        clkinstopped_unused;

  PLLE2_ADV
  #(.BANDWIDTH            ("OPTIMIZED"),
    .COMPENSATION         ("ZHOLD"),
    .STARTUP_WAIT         ("FALSE"),
    .DIVCLK_DIVIDE        (1),
    .CLKFBOUT_MULT        (9),
    .CLKFBOUT_PHASE       (0.000),
    .CLKOUT0_DIVIDE       (9),
    .CLKOUT0_PHASE        (0.000),
    .CLKOUT0_DUTY_CYCLE   (0.500),
    .CLKIN1_PERIOD        (10.000))
  plle2_adv_inst
    // Output clocks
   (
    .CLKFBOUT            (clkfbout_clk_wiz_0),
    .CLKOUT0             (clk_out1_clk_wiz_0),
    .CLKOUT1             (clkout1_unused),
    .CLKOUT2             (clkout2_unused),
    .CLKOUT3             (clkout3_unused),
    .CLKOUT4             (clkout4_unused),
    .CLKOUT5             (clkout5_unused),
     // Input clock control
    .CLKFBIN             (clkfbout_buf_clk_wiz_0),
    .CLKIN1              (clk_in1_clk_wiz_0),
    .CLKIN2              (1'b0),
     // Tied to always select the primary input clock
    .CLKINSEL            (1'b1),
    // Ports for dynamic reconfiguration
    .DADDR               (7'h0),
    .DCLK                (1'b0),
    .DEN                 (1'b0),
    .DI                  (16'h0),
    .DO                  (do_unused),
    .DRDY                (drdy_unused),
    .DWE                 (1'b0),
    // Other control and status signals
    .LOCKED              (locked_int),
    .PWRDWN              (1'b0),
    .RST                 (1'b0));

  assign locked = locked_int;

  BUFG clkf_buf
   (.O (clkfbout_buf_clk_wiz_0),
    .I (clkfbout_clk_wiz_0));

  BUFG clkout1_buf
   (.O   (clk_out1),
    .I   (clk_out1_clk_wiz_0));

endmodule
```

When trying to obtain a valid bitstream using the following command line `nextpnr-xilinx --chipdb ../../chipdb/xc7a35tcsg324.bin --xdc blinky.xdc --json blinky.json --fasm blinky.fasm` I obtain the following error:

```text
Info: Running post-routing legalisation...
terminate called after throwing an instance of 'nextpnr_xilinx::assertion_failure'
  what():  Assertion failure: unsupported compensation type (/build/source/xilinx/fasm.cc:1549)
make: *** [../openXC7.mk:40: blinky.fasm] Aborted
make: *** Deleting file 'blinky.fasm'
```

Standard blinky example works and [litex-ddr-arty-s7](https://github.com/openXC7/demo-projects/tree/main/litex-ddr-arty-s7) also implements.

Any help would be appreciated!!

exitrip, 2024-05-04:
Also happens with xc7a50t based project which builds successfully in Vivado.
skipping a bunch of wire and input buffer generated by clk_wiz:

```
PLLE2_ADV
  #(.BANDWIDTH            ("OPTIMIZED"),
    .COMPENSATION         ("ZHOLD"),
    .STARTUP_WAIT         ("FALSE"),
    .DIVCLK_DIVIDE        (1),
    .CLKFBOUT_MULT        (40),
    .CLKFBOUT_PHASE       (0.000),
    .CLKOUT0_DIVIDE       (10),
    .CLKOUT0_PHASE        (0.000),
    .CLKOUT0_DUTY_CYCLE   (0.500),
    .CLKOUT1_DIVIDE       (8),
    .CLKOUT1_PHASE        (0.000),
    .CLKOUT1_DUTY_CYCLE   (0.500),
    .CLKIN1_PERIOD        (40.000))
  plle2_adv_inst
    // Output clocks
   (
    .CLKFBOUT            (clkfbout_clk_wiz_0),
    .CLKOUT0             (clk_out1_clk_wiz_0),
    .CLKOUT1             (clk_out2_clk_wiz_0),
    .CLKOUT2             (clkout2_unused),
    .CLKOUT3             (clkout3_unused),
    .CLKOUT4             (clkout4_unused),
    .CLKOUT5             (clkout5_unused),
     // Input clock control
    .CLKFBIN             (clkfbout_buf_clk_wiz_0),
    .CLKIN1              (clk_in1_clk_wiz_0),
    .CLKIN2              (1'b0),
     // Tied to always select the primary input clock
    .CLKINSEL            (1'b1),
    // Ports for dynamic reconfiguration
    .DADDR               (7'h0),
    .DCLK                (1'b0),
    .DEN                 (1'b0),
    .DI                  (16'h0),
    .DO                  (do_unused),
    .DRDY                (drdy_unused),
    .DWE                 (1'b0),
    // Other control and status signals
    .LOCKED              (locked_int),
    .PWRDWN              (1'b0),
    .RST                 (1'b0));

// Clock Monitor clock assigning
//--------------------------------------
 // Output buffering
  //-----------------------------------

   BUFG clkf_buf
    (.O (clkfbout_buf_clk_wiz_0),
     .I (clkfbout_clk_wiz_0));
//assign clkfbout_buf_clk_wiz_0 = clkfbout_clk_wiz_0;

   BUFG clkout1_buf
    (.O   (clk_out1),
     .I   (clk_out1_clk_wiz_0));
//assign clk_out1 = clk_out1_clk_wiz_0;

   BUFG clkout2_buf
    (.O   (clk_out2),
     .I   (clk_out2_clk_wiz_0));
//assign clk_out2 = clk_out2_clk_wiz_0;
```

exitrip, 2024-05-04:
Changing to `.COMPENSATION         ("INTERNAL"), //("ZHOLD"),` does allow PNR to continue, and the rest of openXC7 produces a bitstream, but the bitstream doesn't function on the chip (for my DVI out project at least).

@jtplaarj Vivado implementation error is pretty clear what the issue is:
```
[DRC REQP-1684] Feedback check: Unsupported PLLE2_ADV connectivity. The signal clk_tmds_i/clkfbout_buf_clk_wiz_0 on the CLKFBIN pin of clk_tmds_i/plle2_adv_inst with COMPENSATION mode 'INTERNAL' must be driven directly by the CLKFBOUT pin on the same cell.
```
I imagine if you re-run the clk_wiz IP to generate a valid "INTERNAL" compensation PLL setup, it has a chance of working with nextpnr.

exitrip, 2024-05-04:
@jtplaarj following the fix suggested by Vivado, I have a working bitstream once I removed the input buffer after changing the PLL compensation type:
```
// Clock Monitor clock assigning
//--------------------------------------
 // Output buffering
  //-----------------------------------

//   BUFG clkf_buf
//    (.O (clkfbout_buf_clk_wiz_0),
//     .I (clkfbout_clk_wiz_0));
assign clkfbout_buf_clk_wiz_0 = clkfbout_clk_wiz_0;
```