gatecat/nextpnr-xilinx #83: Router2 fails to converge on XC7A100 and XC7A200

cestrauss, 2024-07-25:
Router2 entered a seemingly infinite loop on one of my designs, for XC7A100. I recursively deleted ranges of cells from the input JSON until I reduced it to just three nets and three cells: two LUT1 and a FDCE in a loop. XC7A200 also shows the problem, but XC7A35 and XC7A50 complete successfully. Router1 doesn't seem to be affected.
Git commit is current "xilinx-upstream" branch, f79387599b420bfadf3dc218e55e1e77b1edb304.
Reduced design: [top.json](https://github.com/user-attachments/files/16383894/top.json)
Shell build script: [build.txt](https://github.com/user-attachments/files/16383934/build.txt)
Log output: [top_tim.txt](https://github.com/user-attachments/files/16383941/top_tim.txt)
Log output (debug): [top_debug_tim.txt](https://github.com/user-attachments/files/16383943/top_debug_tim.txt)

lehaifeng000, 2024-08-05:
why there is 2 LUT1 ? I think it should be optimized by abc9...

lehaifeng000, 2024-08-05:
and there is no iopad and buf?

cestrauss, 2024-08-10:
Sorry, it was not created by Yosys, it was hand edited by me.

Here follows the full design, which really was created by Yosys:  [top.zip](https://github.com/user-attachments/files/16570882/top.zip)

It fails to converge with Router2.